Severson-Group/docs.amdc.dev #18: Challenge with Programming AMDC instructions for dual core

elsevers, 2022-01-07:
I am struggling with [these instructions](https://docs.amdc.dev/firmware/xilinx-tools/building-and-running-firmware.html#programming-amdc):

[image]

**Step 1: Following the instructions so far, I don't have an `app_cpu0` project:**

[image]

`Attention` block: when I attempt to browse for an ELF file for the `cpu0` I am not finding it in my `Debug/` subfolder:
[image]

[image]

Did I do something wrong? Do the directions need to be adjusted?

npetersen2, 2022-02-11:
You did not add the `app_cpu0` project to your SDK environment, so it did not get compiled, so there is no elf file

elsevers, 2022-02-11:
Thanks. I guess I am confused. When I go through [these instructions](https://docs.amdc.dev/firmware/xilinx-tools/building-and-running-firmware.html), where should I have set up the `app_cpu0` project?

npetersen2, 2022-02-11:
`app_cpu0` should be added at this part:

https://docs.amdc.dev/firmware/xilinx-tools/building-and-running-firmware.html#import-projects-into-sdk

elsevers, 2022-02-11:
Thanks. Should `app_cpu0` project have been moved over to my private user code folder or should it remain in the `AMDC-Firmware` submodule?

PR #24 (2022-02-13): Add app_cpu0 to building and running firmware instructions
The `Building and Running Firmware` instructions currently does not have the users add the `app_cpu0` project when following the `Private User Applications` pathway. This fixes #18. 

This PR updates the Building and Running Firmware` instructions to resolve this. (@npetersen2, have I correctly understood the intended workflow in these instructions?))
closed 2022-02-13 by npetersen2