Severson-Group/docs.amdc.dev #60: Document instructions for copying Xilinx files or projects

wknutson, 2022-07-07:
## Copying Xilinx Files/Projects
When copying files from one Xilinx project to another, they may not show up in the Project Explorer after importing via File -> Open Projects from File System. Solution 1 relies on you to delete a `.project` file and Xilinx to regenerate it. Solution 2 seems to overwrite `.project` files. 

## Solutions
#### Solution 1: 
This can be fixed by Navigating to the folder you want to import and deleting the `.project` file found within that folder. Now it should import via File -> Open Projects from File System. 

#### Solution 2: 
Another solution is to start in Vivado. After getting a successful block diagram and generating a bitstream, select File -> Export -> Export Hardware. Leave 'Include Bitstream' unchecked, and select the folder that contains the code you want to modify (If you are following these instructions, https://docs.amdc.dev/firmware/xilinx-tools/building-and-running-firmware.html#fix-common-code-compilation, `my-AMDC-private-C-code` is the folder that you should select). Click OK, and select File -> Launch SDK. Your Exported Location should be the same folder as before (`my-AMDC-private-C-code`), and your workspace should be the `SDK` folder found inside `AMDC-Firmware`. Click OK and close the welcome tab in Xilinx when it loads. There should only be `amdc_rev*_wrapper_hw_platform_0` in your Project Explorer. Now you can import files from the folders you selected earlier (`my-AMDC-private-C-code` and `SDK`) via File -> Open Projects from File System.

npetersen2, 2022-07-24:
@wknutson thanks for documenting these changes!

Can you create a PR with these changes for review? Since these look like text-only changes, try just updating the files via the GitHub interface. You'll need to make a user branch with your changes, then create a PR for review and merging.

If you are interested in building the docs locally so you can see your changes rendered, see #56

codecubepi, 2024-08-20:
Looks like this was resolved in #63

closed 2024-08-20 by codecubepi