Severson-Group/docs.amdc.dev #67: Broken links on Firmware Architecture page

elsevers, 2022-11-27:
The following links are broken on the [Firmware Architecture](https://docs.amdc.dev/firmware/arch/index.html) page:
1. `Read more about tasks...` link in the [tasks](https://docs.amdc.dev/firmware/arch/index.html#tasks) section 
2. `Read more about commands...` link in the [commands](https://docs.amdc.dev/firmware/arch/index.html#commands) section

npetersen2, 2022-11-28:
It is not immediately obvious to me how to fix this, but some ideas from here might help:

https://stackoverflow.com/questions/15394347/adding-a-cross-reference-to-a-subheading-or-anchor-in-another-page

codecubepi, 2024-08-20:
@npetersen2 

I'm going through the backlog of issues in the docs repo and found this... seems to be working for me? Maybe it was fixed at some point? I think we can close this one unless you find otherwise that this is still a problem.

npetersen2, 2024-08-20:
Thanks @codecubepi , I agree, think looks to be solved already.

closed 2024-08-20 by npetersen2