Severson-Group/docs.amdc.dev #98: Update docs for new AMDS Interface

codecubepi, 2024-05-21:
Most updates probably just need to be made to the AMDS > Firmware page ([src](https://github.com/Severson-Group/docs.amdc.dev/blob/main/source/accessories/amds/firmware/index.md) and [live page](https://docs.amdc.dev/accessories/amds/firmware/index.html)), but I'll do an all-file search in the docs repo for references to the AMDS to make sure everything is up to date.

npetersen2, 2024-05-28:
Make sure to also update this page: https://docs.amdc.dev/getting-started/user-guide/amds-interface.html

codecubepi, 2024-05-30:
Okay, I think this is probably good now. Please take a look at #99 again.

codecubepi, 2024-06-03:
Closing with approval of #99 (even though it won't be merged right away).

PR #99 (2024-09-13): Update AMDS Firmware page for new updates made in AMDC v1.3
Resolves #98 

I updated any pages about the AMDS that had outdated information, whether it be changes in AMDC Firmware v1.3, or older.
closed 2024-06-03 by codecubepi