Severson-Group/docs.amdc.dev #100: Update docs for new timing manager operation and configuration

npetersen2, 2024-05-28:
TODO

Known4225, 2024-10-18:
In 10/18 meeting with @elsevers , we are thinking we'll do the following for our tutorial 1 so that we don't even require the user to have a uInverter:

1. One really simple tutorial on using AMDC timing manager + task + ADC and no additional hardware. Outline:
    - Story/theory:
        - Introduce goals of the timing manager from the perspective of VSI + hardware command for changing PWM frequency and wanting to synchronize 1) data collection to start at a value/peak; 2) only running a task once all data is collected.
        - scheduling your task for data inputs:
            - Synchronized vs legacy mode. Explain that this tutorial is covering the synchronized mode.
            - timing manager event ratio, pwm switching frequency, what happens when the us in header file is mismatched to the event ratio: not an integer multiple, too slow, faster).   
    - Code it: Modify VSI tutorial code to 
        - Configure task with add timing_manager_enable_sensor(ADC)
        - Set up ADC to read channel xxx
        - Set `Do` to be `abs(Vin / 12V)`
        - Enhance `stats` command: by using the timing manager diagnostics commands to display
            - Already present: Loop and run time (stats on the orange `control task running` block)
            - Sensor sampling window (duration of the gray `sensors sampling` block and loop timing between consecutive sensor sampling windigs)
        - Add serial port command to change the ADC clock rate
    - Demonstrate it
        1. Tell user it is optional to: Use multimeter or oscilloscope to measure output of PWM port. Can optionally attach a variable dc supply to ADC pin and change ratio
        2. with default ADC clock and PWM frequency, use `stats` command to measure the `sensors sampling` and `control task running` blocks
        3. change ADC clock rate and remeasure
        4. change PWM frequency and remeasure (demonstrate two cases: 1) where the task keeps its same rate 2) where new PWM frequeny causes task to run less frequently because the task period is not a multiple of the PWM period)
        
 Not yet spec'd out, but we plan to add a more advanced tutorial along the lines of what is described above with the AMDC + uInverter and all the different options.

Known4225, 2024-10-30:
First draft complete in branch [user/known4225/tm-tutorials](https://github.com/Severson-Group/docs.amdc.dev/tree/user/known4225/tm-tutorials). Has a series of three experiments which change the timing manager event ratio as well as the pwm frequency to observe the changes to task rate (loop time). Uses a dummy ADC sensor (the value of which is not used) to enable post-sensor mode on the timing manager (to observe effects on control task timing).

elsevers, 2024-10-31:
Copy-paste of my feedback from Slack:

Let's get this set up as a PR requesting to merge into `user/codecubepi/tm-sched-updates` so that Patrick and I can provide reviews on it formally.

# Summary
My main feedback right now is that it is a pretty hard to read it and understand what to do.

## Typesetting / Formatting

Some of this is typesetting/formatting. Like, these snips of equations are quite hard to read, and especially when some of the parts of the equation are appearing as plain text instead of inside a code environment.
[image]

[image]

[image]

Take a look at how Nathan renders equations in the VSI tutorial for an example on how to make things more readable

[image]

## Structure
A big part of this is structure. It isn't clear when the text or the headings when we are supposed to take action and code something vs when you are giving us background and theory. See some of the other tutorials as examples for structure that helps with conveying instruction

## Perspective
I think the biggest thing though is that the perspective is a bit off.

1. We should be entering this tutorial fresh off of the previous tutorials that you linked and told that now we are going to revise this user app that was already created so that it's execution will be synchronized with the completion of the sensor data acquisition and that this is important for precision motor controls.
2. We should then get a brief background on the way the user app was previously scheduled by the OS (legacy) and a brief explanation on that now it will be scheduled based on completion of data measurement (post-sensor). Tell us a bit about how PWM factors into this, etc., but hold off on the heavy details for later (or refer uses to parts o the docs page on it).
3. Then I think you should guide us through a series of steps that explicitly have us change the VSI code to prepare for the experiments. When you do this, you can sprinkle in a bit more detail on theory as the user thinks about the code they are adding and the parameters that they set. Nathan was the master at doing this in his tutorials. (have the user do some work and then do some thinking and then some more work, etc.)
4. Then guide us through a series of steps of the experiment and thinking about the results.

(this list is consistent with [this comment above](https://github.com/Severson-Group/docs.amdc.dev/issues/100#issuecomment-2422736235)

# Next Steps
Let's have you go ahead and revise, and then create a PR and add @codecubepi and myself as reviewers

elsevers, 2024-11-06:
Feedback:
1. Add a sentence within `Tutorial Requirements`, after the list, telling us that this tutorial assumes all new code is being added to the code created in tutorials listed above.

PR #112 (2024-12-20): Timing Manager Tutorial
This PR closes #100.

This Timing & Sensors tutorial expands upon the VSI tutorial and enables the ADC sensor to observe the effects of the timing manager. Users will learn how certain tuning parameters (mainly the User Event Ratio and the PWM frequency) can affect the loop time of their control tasks.
closed 2025-01-02 by elsevers