Severson-Group/docs.amdc.dev #127: Update AMDS sensor card docs to provide an expression for the sensed quantity in terms of the ADC voltage

elsevers, 2025-02-28:
# Abstract
We want each of the AMDS sensor card docs to provide an expression for the sensed quantity (i.e., current in amps for the current card) in terms of the ADC voltage.

# Context

The [AMDS current card](https://docs.amdc.dev/accessories/amds/sensor-cards/current/index.html#final-primary-current-to-adc-input-voltage-relationship) currently does the opposite of what this issue is requesting (it provides the ADC terminal voltage in terms of the sensed quantity).

# Approach

Add to each of the AMDS card docs a section called `### Relationship Between Measured Signal and `

# This is a draft and @elsevers will finish editing later

Daehoon-Sung, 2025-03-05:
### Current Card
For REV B,

$$
 I_{\text{PRIMARY}} = (V_{\text{ADC}}-2.4922) \times 29.4118  [A]
$$

For REV C,

$$
 I_{\text{PRIMARY}} = (V_{\text{ADC}}-2.5126) \times 29.4118  [A]
$$

### Low voltage Card

$$
V_{\text{meas}} = V_{\text{IN+}} - V_{\text{IN-}} =  (V_{\text{ADC}} - 2.048) \times 10  [V]
$$

### high Voltage Card
 will be updated

elsevers, 2025-03-06:
Thanks @Daehoon-Sung!

Known4225, 2025-03-21:
The offset value for the low voltage card should be 0 instead of 2.048

[image]

This is because the ADC reports the measured voltage relative to the 2.048V reference

Daehoon-Sung, 2025-11-04:
Hello Professor @elsevers Dr @noguchi-takahiro . I think we need to huddle up together to review this one. 
Do you prefer to review this during the office hour or do you want to setup a specific time for this?